gotzl/hid-fanatecff #109: Fanatec BMW GT2 wheel - missing analog binding/axis.

kropop, 2024-11-23:
Hello,

I've recently tried the Fanatec driver under Linux using a native version of Euro Truck Simulator 2. I could bind all buttons and axis from the rest of my controls - all pedals (V3), the shifter and the wheel controls are bindable and work well, except for the left analog stick present on the GT2 wheel. Is there a workaround for this?

The Proton version of ETS2 doesn't detect that input either.

gotzl, 2024-12-08:
This sounds similar to #77 
Which version of the driver do you use?

kropop, 2024-12-12:
Hello

I used the most recent version of the driver.

I tried evtest on the wheel and indeed, it seems like the axis values are all wrong:
```
Event: time 1734039298.479945, type 3 (EV_ABS), code 3 (ABS_RX), value -16
Event: time 1734039298.479945, -------------- SYN_REPORT ------------
Event: time 1734039299.346047, type 3 (EV_ABS), code 3 (ABS_RX), value 18
Event: time 1734039299.346047, -------------- SYN_REPORT ------------
Event: time 1734039300.557188, type 3 (EV_ABS), code 4 (ABS_RY), value 2
Event: time 1734039300.557188, -------------- SYN_REPORT ------------
Event: time 1734039301.006241, type 3 (EV_ABS), code 4 (ABS_RY), value -30
Event: time 1734039301.006241, -------------- SYN_REPORT ------------
```

That was me moving the stick all the way to the left and right, then up and down. However, running the command mentioned in the issue you linked does not in fact callibrate the axes correctly for me, despite axis 3 and 4 being listed as ABS_RX and ABS_RY for me. The wheel is confirmed as working under Windows. Could it be an issue with my firmware?

gotzl, 2024-12-12:
Can you check/post the calibration with `evdev-joystick -s /dev/input/by-id/usb-Fanatec*-event-joystick`?
Did only these 4 values appear when you were moving the stick? That's very odd...

SynthetikzZ, 2024-12-13:
The analog stick isnt mappable for a couple of games on Windows too. It's a problem on fanatecs side.

kropop, 2024-12-15:
> Can you check/post the calibration with `evdev-joystick -s /dev/input/by-id/usb-Fanatec*-event-joystick`? Did only these 4 values appear when you were moving the stick? That's very odd...

```
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 32782, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
  Absolute axis 0x01 (1) (Y Axis) (value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
  Absolute axis 0x02 (2) (Z Axis) (value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
  Absolute axis 0x03 (3) (X Rate Axis) (value: 0, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 0, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
  Absolute axis 0x06 (6) (Throttle) (value: 255, min: 0, max: 255, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x07 (7) (Rudder) (value: 0, min: -128, max: 127, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
```

If I wiggle the stick, more values show up, but they only show up at the very extremes of the range of the stick:

```
Event: time 1734219830.452255, -------------- SYN_REPORT ------------
Event: time 1734219831.046327, type 3 (EV_ABS), code 4 (ABS_RY), value 6
Event: time 1734219831.046327, -------------- SYN_REPORT ------------
Event: time 1734219831.050328, type 3 (EV_ABS), code 3 (ABS_RX), value 31
Event: time 1734219831.050328, -------------- SYN_REPORT ------------
Event: time 1734219832.251468, type 3 (EV_ABS), code 3 (ABS_RX), value -1
Event: time 1734219832.251468, -------------- SYN_REPORT ------------
Event: time 1734219832.272470, type 3 (EV_ABS), code 3 (ABS_RX), value -32
Event: time 1734219832.272470, -------------- SYN_REPORT ------------
Event: time 1734219832.616514, type 3 (EV_ABS), code 3 (ABS_RX), value 7
Event: time 1734219832.616514, -------------- SYN_REPORT ------------
Event: time 1734219834.289716, type 3 (EV_ABS), code 3 (ABS_RX), value -25
Event: time 1734219834.289716, -------------- SYN_REPORT ------------
Event: time 1734219835.210823, type 3 (EV_ABS), code 3 (ABS_RX), value 9
Event: time 1734219835.210823, -------------- SYN_REPORT ------------
Event: time 1734219835.782902, type 3 (EV_ABS), code 3 (ABS_RX), value -24
Event: time 1734219835.782902, -------------- SYN_REPORT ------------
Event: time 1734219836.257952, type 3 (EV_ABS), code 4 (ABS_RY), value -27
Event: time 1734219836.257952, -------------- SYN_REPORT ------------
Event: time 1734219836.642996, type 3 (EV_ABS), code 4 (ABS_RY), value 5
Event: time 1734219836.642996, -------------- SYN_REPORT ------------
Event: time 1734219838.365209, type 3 (EV_ABS), code 3 (ABS_RX), value 9
Event: time 1734219838.365209, -------------- SYN_REPORT ------------
Event: time 1734219838.876269, type 3 (EV_ABS), code 3 (ABS_RX), value -24
Event: time 1734219838.876269, -------------- SYN_REPORT ------------
Event: time 1734219839.419332, type 3 (EV_ABS), code 3 (ABS_RX), value 13
Event: time 1734219839.419332, -------------- SYN_REPORT ------------
Event: time 1734219842.084654, type 3 (EV_ABS), code 3 (ABS_RX), value -20
Event: time 1734219842.084654, -------------- SYN_REPORT ------------
Event: time 1734219842.560705, type 3 (EV_ABS), code 3 (ABS_RX), value 15
Event: time 1734219842.560705, -------------- SYN_REPORT ------------
Event: time 1734219843.729846, type 3 (EV_ABS), code 3 (ABS_RX), value -17
Event: time 1734219843.729846, -------------- SYN_REPORT ------------
Event: time 1734219844.168908, type 3 (EV_ABS), code 3 (ABS_RX), value 16
Event: time 1734219844.168908, -------------- SYN_REPORT ------------
Event: time 1734219844.174910, type 3 (EV_ABS), code 4 (ABS_RY), value -28
Event: time 1734219844.174910, -------------- SYN_REPORT ------------
Event: time 1734219844.590949, type 3 (EV_ABS), code 3 (ABS_RX), value -16
Event: time 1734219844.590949, -------------- SYN_REPORT ------------
Event: time 1734219846.217145, type 3 (EV_ABS), code 3 (ABS_RX), value 19
Event: time 1734219846.217145, -------------- SYN_REPORT ------------
Event: time 1734219846.942234, type 3 (EV_ABS), code 3 (ABS_RX), value -12
Event: time 1734219846.942234, -------------- SYN_REPORT ------------
Event: time 1734219849.044487, type 3 (EV_ABS), code 4 (ABS_RY), value 4
Event: time 1734219849.044487, -------------- SYN_REPORT ------------
Event: time 1734219849.850584, type 3 (EV_ABS), code 4 (ABS_RY), value -29
Event: time 1734219849.850584, -------------- SYN_REPORT ------------
Event: time 1734219850.532666, type 3 (EV_ABS), code 4 (ABS_RY), value 2
Event: time 1734219850.532666, -------------- SYN_REPORT ------------
Event: time 1734219851.291763, type 3 (EV_ABS), code 4 (ABS_RY), value -30
Event: time 1734219851.291763, -------------- SYN_REPORT ------------
[... 7 lines cut ...]
`evdev-joystick --evdev /dev/input/by-id/usb-Fanatec_*-event-joystick --axis 3 --min -128 --max 128 --axis 4 --min -128 --max 128` 
and then
`evdev-joystick -s /dev/input/by-id/usb-Fanatec*-event-joystick` command again produces these results instead:

```
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 32764, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 255)
[... cut ...]
It seems like it does not detect the entire range of the movement of the stick, as moving the steering wheel little by little produces a lot of messages, while wiggling the stick into extremes causes very few event readings.

> The analog stick isnt mappable for a couple of games on Windows too. It's a problem on fanatecs side.

It is mappable in the games that I'm trying in Windows. I know ACC has an input for "look left/right" and only accepts buttons, but in ETS2 and ATS there is a mappable look axis.

Side question - is there any reason as to why I have to reset my deadzone for the device every time I start it up?

gotzl, 2024-12-15:
Hi,

the problem here seems to be the `fuzz: 255`. Try the following commands:
```
evdev-joystick --evdev /dev/input/by-id/usb-Fanatec_*-event-joystick --axis 3 --min -128 --max 128 --fuzz 0
evdev-joystick --evdev /dev/input/by-id/usb-Fanatec_*-event-joystick --axis 4 --min -128 --max 128 --fuzz 0
```
Note that you've to execute the command for each axis.
I wonder why the `fuzz` isn't set to zero to begin with. There is the udev rule that should take care of that, see https://github.com/gotzl/hid-fanatecff/blob/4f80db5d4dd776e7ebe9ac8b0065d0b06baab77a/fanatec.rules#L16

> Side question - is there any reason as to why I have to reset my deadzone for the device every time I start it up?

You shouldn't have to. There is the udev rule that should take care of that. Check the output of
`udevadm test /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick` and see if `/etc/udev/rules.d/99-fanatec.rules` gets called.

kropop, 2024-12-15:
> You shouldn't have to. There is the udev rule that should take care of that. Check the output of
> `udevadm test /dev/input/by-id/usb-Fanatec_FANATEC_Wheel-event-joystick` and see if `/etc/udev/rules.d/99-fanatec.rules` gets called.

there is a `Reading rules file: /etc/udev/rules.d/99-fanatec.rules` line in there, and later on
```
event26: /usr/local/lib/udev/rules.d/99-fanatec-wheel-perms.rules:7 MODE 0666
event26: /etc/udev/rules.d/99-fanatec.rules:5 MODE 0666

```
> Note that you've to execute the command for each axis.

It did fix that issue indeed, I'm getting a lot of values now read by the stick and I can bind them in ATS and ETS2 now.
Question is now, why are udev rules not working for me. The udev rules I have set for my T-LCM pedals are working okay and are loaded every time I plug the device in.

I went ahead and reinstalled the driver again, copying the github repo and all. Restarted my PC and udev rules *seem* to be working now - both the wheel deadzone and the stick are fine.

kropop, 2024-12-21:
A recent update to my system broke this again, and also reset the udev rules for the rest of the wheel. I'm starting to believe something makes the udev rules break during a full update. Weirdly enough, the ones I have set up for my Thrustmaster pedals seem to be untouched, but I had to reinstall the driver yet again - not only the stick axis was borked, so was the deadzone for main steering axis.

gotzl, 2024-12-22:
> reset the udev rules for the rest of the wheel

What do you mean by that? Is the .rules file removed by your system update? What system is it?

gotzl, 2025-01-03:
Is there another rule afterwards that does s.t. to the device? Or is there some service running that tries to apply some 'calibration' for the device?

kropop, 2025-01-04:
What I can see from udevadm test is that these two files get called:
```
event26: /usr/local/lib/udev/rules.d/99-fanatec-wheel-perms.rules:7 MODE 0666
event26: /etc/udev/rules.d/99-fanatec.rules:5 MODE 0666
```

I don't have any services that could in theory mess with the wheel. 

What I've noticed is that each time this happens, when launching ETS2 it pops up with a message saying that my wheel is missing, and then I have to select a "different" wheel. My axis binding return but they work as if I had no udev rules set. It's as if the system was forgetting about that device altogether.

If that is of any help, I do connect my wheel through a powered USB hub.
I'm attaching my full udevadm test log - the wheel is working, but the axis settings are screwed up. The udev file is called. I have not modified my rules file in any way.
[udevadm test log.txt](https://github.com/user-attachments/files/18305186/udevadm.test.log.txt)

kropop, 2025-01-04:
Another message because I think I'm starting to figure out why this happens.

The Fanatec module gets installed every time into a new kernel directory:
[image]
I'm beginning to suspect that it breaks every time I update my kernel, because right now I'm on version 6.12.6-200 and I have three file copies for each time I had to reinstall the kernel module.

gotzl, 2025-01-08:
Yes, looks like the reason for all this are the kernel updates. Unfortunately, out-of-tree modules like this one have to be compiled/installed again specifically for every kernel. Also, after you've recompile/installed the module, you've to reconnect (or power-cycle) the wheel for the module to get loaded (or load it by hand with `insmod hid-fanatec.ko`).

In the udevadm log you've posted one can see that the rule is there, but it actually doesn't do anything. Most likely because the hid-fanatec module wasn't loaded for the device.

udevadm output should contain a line like this:
```
event28: /etc/udev/rules.d/99-fanatec.rules:18 RUN '/bin/sh -c '/usr/bin/evdev-joystick ...
```

and at the bottom 
```
Queued commands:
  RUN{builtin} : uaccess
  RUN{program} : /bin/sh -c '/usr/bin/evdev-joystick ...
```

kropop, 2025-01-17:
Closing this as explained/solved.

Thank you for your continued work on making simracing on Linux a possibility!

closed 2025-01-17 by kropop