gotzl/hid-fanatecff #110: Sysfs files for LEDs missing

gotzl, 2025-01-30:
I tried this branch as well, after the current tagged release kept crashing the games on startup for me.
And the version in this branch is working great for me so far. No crash yet, at least no wheel related one.

I have a Podium DD1 with a Clubsport F1 2020 wheel (wheel_id: 0x2e), for context.

My question is, if any of the LEDs on the wheel should work with this current version? Because for me the LEDs and the Display on the wheel don't work in ASM2 or F1 22. Could only test those two so far.

While trying the hid-fanatecff-tools, I received an error, that my wheel does not have the leds directory. I checked and it does not have that directory.
The error message was:

```
FileNotFoundError: [Errno 2] No such file or directory: '/sys/module/hid_fanatec/drivers/hid:fanatec/0003:0EB7:0006.001C/leds/0003:0EB7:0006.001C::RPM1/brightness'
```

I'm wondering, if this error is related to my specific wheel or if the current changes no longer provide the .../leds directory.

My dmesg output is:
```
[32310.568401] usb 1-3.1.2.1.4: new full-speed USB device number 25 using xhci_hcd
[32310.674046] usb 1-3.1.2.1.4: New USB device found, idVendor=0eb7, idProduct=0006, bcdDevice= 4.03
[32310.674051] usb 1-3.1.2.1.4: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[32310.674054] usb 1-3.1.2.1.4: Product: FANATEC Podium Wheel Base DD1
[32310.674057] usb 1-3.1.2.1.4: Manufacturer: Fanatec
[32310.747200] hid_map_usage: 34 callbacks suppressed
[32310.747202] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 768 type 1
[32310.747205] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 769 type 1
[32310.747206] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 770 type 1
[32310.747208] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 771 type 1
[32310.747209] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 772 type 1
[32310.747210] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 773 type 1
[32310.747211] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 774 type 1
[32310.747212] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 775 type 1
[32310.747214] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 776 type 1
[32310.747215] Fanatec FANATEC Podium Wheel Base DD1: Invalid code 777 type 1
[32310.747271] input: Fanatec FANATEC Podium Wheel Base DD1 as /devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.1/usb1/1-3/1-3.1/1-3.1.2/1-3.1.2.1/1-3.1.2.1.4/1-3.1.2.1.4:1.0/0003:0EB7:0006.001C/input/input47
[32310.797602] fanatec 0003:0EB7:0006.001C: input,hiddev103: USB HID v1.11 Joystick [Fanatec FANATEC Podium Wheel Base DD1] on usb-0000:08:00.1-3.1.2.1.4/input0
[32310.797924] fanatec 0003:0EB7:0006.001D: hidraw12: USB HID v1.11 Joystick [Fanatec FANATEC Podium Wheel Base DD1] on usb-0000:08:00.1-3.1.2.1.4/input0
[32310.801374] fanatec 0003:0EB7:0006.001C: Hires timer: period = 2 ms
```

_Originally posted by @Nivalux in https://github.com/gotzl/hid-fanatecff/issues/86#issuecomment-2622533983_

gotzl, 2025-01-30:
@Nivalux The kernel configuration determins if the driver is build with `LED` sysfs support, and you've already found `#define CONFIG_LEDS_CLASS`. It seems your kernel was build without leds-class-support, which is abit puzzling since you say it works when setting the define by hand?

Nivalux, 2025-01-31:
@gotzl That was the missing link. I didn't realize, that the CONFIG_LEDS_CLASS flag was a kernel option.
I checked my running kernel and it has that option set to module (=m). The module was also loaded and is in use.

What didn't work on my end was, that I don't know how to pass my current running kernel config to the makeFlags.
I'm using Nix, which probably does this differently than other systems.
I'm now adding that flag in my script, so it build correctly now withouth source code modification.
I now pass `EXTRA_CFLAGS=-DCONFIG_LEDS_CLASS` to `make` and that works.

Maybe a hint could be added to the README, that in the case the LEDs are not working, to check if the kernel was build with that flag set and to enable it manually for the make process, if needed.

Nivalux, 2025-01-31:
About the custom proton version, no I am currently testing with GEProton 9-20. 
That version was released after the patch got merged into wine, so I wanted to see, if it already was included there.
Could not yet test, if one of the F1 games works with it, since I get the denovu error of to many devices and have to wait.
AMS2 worked for me (both FFB and Display+LEDs) with the UDP protocol enabled and using hid-fanatecff-tools.

Nivalux, 2025-01-31:
Update:
I can launch F1 22 and could test, if the LEDs work with GEProton 9-20. They don't.
So I will have to look into how I can build and use your custom proton version.

gotzl, 2025-01-31:
Hi, 

> .. after the patch got merged into wine ...

You mean the patch to wine I've referenced in the PR? Unfortunatelly, proton uses their own wine-fork that doesn't have this patch yet. I've yet to submit it so that it lands in wine-proton.
Using my proton build should be straight forward, just unpack it to the `compatibilitytools.d` as you did with GEProton.

gotzl, 2025-01-31:
> What didn't work on my end was, that I don't know how to pass my current running kernel config to the makeFlags.
I'm using Nix, which probably does this differently than other systems.

Out-of-tree kernel modules a build within the kernel-src directory, see https://github.com/gotzl/hid-fanatecff/blob/399c6c42606d864ce9235738426d035b09c1e750/Makefile#L8 
Question is, what Nix provides in the kernel-src dir ...

Nivalux, 2025-01-31:
It does provide the correct path there, I think.
The log output looks like this:
`build flags: SHELL=/nix/store/gwgqdl0242ymlikq9s9s62gkp5cvyal3-bash-5.2p37/bin/bash KVERSION=6.12.10 KERNEL_SRC=/nix/store/lml00n1wlpsjdhnwqd13zqai2flanj1w-linux-6.12.10-dev/lib/modules/6.12.10/build MODULEDIR=\$\(out\)/lib/modules/6.12.10/kernel/drivers/hid`

and the content of the KERNEL_SRC dir is:
```
dr-xr-xr-x 3 root root       3  1. Jan 1970  arch
dr-xr-xr-x 4 root root       4  1. Jan 1970  include
dr-xr-xr-x 2 root root       4  1. Jan 1970  kernel
-r--r--r-- 2 root root     245  1. Jan 1970  Makefile
-r--r--r-- 2 root root 2292645  1. Jan 1970  Module.symvers
dr-xr-xr-x 5 root root      38  1. Jan 1970  rust
dr-xr-xr-x 7 root root      18  1. Jan 1970  scripts
lrwxrwxrwx 2 root root       9  1. Jan 1970  source -> ../source
dr-xr-xr-x 4 root root       4  1. Jan 1970  tools
```

On another note, I also used your custom proton version now. Since it crashes for the F1 games I could only test it with ACC so far. I could not notice a difference between it and the GE Proton version. The Display and Leds didn't work.
Do I need to reset something in ACC for the wheel? I think you wrote, that it uses the fanatec SDK and should have working LEDs?

Nivalux, 2025-01-31:
nvm, I found the option for Fanatec LEDs in ACC. It works!
LEDs and even the rumble from the peddals works, awesome!

gotzl, 2025-01-31:
Great to hear it works for you in ACC now. The F1 games don't work for me either. Not sure what's going on there ....

Does `grep CONFIG_LEDS_CLASS /nix/store/lml00n1wlpsjdhnwqd13zqai2flanj1w-linux-6.12.10-dev/lib/modules/6.12.10/build/include/generated/autoconf.h` output anything?

Nivalux, 2025-02-01:
That was the missing piece in the puzzle. The output is:
```
#define CONFIG_LEDS_CLASS_MULTICOLOR_MODULE 1
#define CONFIG_LEDS_CLASS_FLASH_MODULE 1
#define CONFIG_LEDS_CLASS_MODULE 1
```

So I changed the header to this and it works on my end:
```
diff --git a/hid-ftec.h b/hid-ftec.h
index 4757659..61ecd21 100644
--- a/hid-ftec.h
+++ b/hid-ftec.h
@@ -6,6 +6,10 @@

 #define FANATEC_VENDOR_ID 0x0eb7

+#ifdef CONFIG_LEDS_CLASS_MODULE
+#define CONFIG_LEDS_CLASS
+#endif
+
 // wheelbases
 #define CLUBSPORT_V2_WHEELBASE_DEVICE_ID 0x0001
 #define CLUBSPORT_V25_WHEELBASE_DEVICE_ID 0x0004
```

Maybe it is safer, to make your own define, like `FANATEC_HAS_LED_SUPPORT`, then it's easier to catch an error with missing includes.

closed 2025-02-01 by gotzl

Nivalux, 2025-02-01:
Thank you! Yes, the issue is fixed.

commit 38be773f (2025-02-01): replace `#ifdef CONFIG_LEDS_CLASS` by `#if IS_REACHABLE(CONFIG_LEDS_CLASS)`, fixes #110