gotzl/hid-fanatecff #111: Make/make install does not work

gotzl, 2025-01-30:
How do I use this driver?
I'd like to try it out for AC Evo:)

Can't install it via make/make install

_Originally posted by @Dekalliver in https://github.com/gotzl/hid-fanatecff/issues/86#issuecomment-2625322616_

samowitsch, 2025-04-20:
I tried it also and ```make``` is not working on Fedora Linux 41 (Workstation Edition).

```bash
csg@fedora:~/git/hid-fanatecff$ make

:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Verzeichnis „/home/csg/git/hid-fanatecff“ wird betreten
make[1]: *** /lib/modules/6.11.4-301.fc41.x86_64/build: Datei oder Verzeichnis nicht gefunden.  Schluss.
make[1]: Verzeichnis „/home/csg/git/hid-fanatecff“ wird verlassen
make: *** [Makefile:8: default] Fehler 2

csg@fedora:~/git/hid-fanatecff$ ll /lib/modules/6.11.4-301.fc41.x86_64/build
lrwxrwxrwx. 1 root root 39 20. Okt 2024  /lib/modules/6.11.4-301.fc41.x86_64/build -> /usr/src/kernels/6.11.4-301.fc41.x86_64
```

samowitsch, 2025-04-20:
I did a check on my non-gaming Ubuntu PC. There ```make``` / ```sudo make install``` works like it should.

```bash
csg@csg-deskmini-300:~/git/hid-fanatecff$ make
-e 
:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-58-generic“ wird betreten
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
  CC [M]  /home/csg/git/hid-fanatecff/hid-ftec.o
  CC [M]  /home/csg/git/hid-fanatecff/hid-ftecff.o
  CC [M]  /home/csg/git/hid-fanatecff/hid-ftecff-tuning.o
  LD [M]  /home/csg/git/hid-fanatecff/hid-fanatec.o
  MODPOST /home/csg/git/hid-fanatecff/Module.symvers
  CC [M]  /home/csg/git/hid-fanatecff/hid-fanatec.mod.o
  LD [M]  /home/csg/git/hid-fanatecff/hid-fanatec.ko
  BTF [M] /home/csg/git/hid-fanatecff/hid-fanatec.ko
Skipping BTF generation for /home/csg/git/hid-fanatecff/hid-fanatec.ko due to unavailability of vmlinux
make[1]: Verzeichnis „/usr/src/linux-headers-6.8.0-58-generic“ wird verlassen

csg@csg-deskmini-300:~/git/hid-fanatecff$ sudo make install
-e 
:: Installing Fanatec kernel module/udev rule
=====================================================
'hid-fanatec.ko' -> '/lib/modules/6.8.0-58-generic/kernel/drivers/hid/hid-fanatec.ko'
'fanatec.rules' -> '/etc/udev/rules.d/99-fanatec.rules'
depmod

csg@csg-deskmini-300:~/git/hid-fanatecff$ sudo udevadm control --reload-rules && sudo udevadm trigger

csg@csg-deskmini-300:~/git/hid-fanatecff$ lsmod | grep fanatec
hid_fanatec            57344  0
hid                   180224  4 usbhid,hid_apple,hid_generic,hid_fanatec
```

samowitsch, 2025-04-20:
On Fedora it seems there is a faulty symlink (sorry my console output is in german).
6.11.4-301.fc41.x86_64  should be 6.13.11-200.fc41.x86_64?

```bash
csg@fedora:~/git/hid-fanatecff$ ll /lib/modules/6.11.4-301.fc41.x86_64/build 
lrwxrwxrwx. 1 root root 39 20. Okt 2024  /lib/modules/6.11.4-301.fc41.x86_64/build -> /usr/src/kernels/6.11.4-301.fc41.x86_64

csg@fedora:~/git/hid-fanatecff$ ll /usr/src/kernels/6.11.4-301.fc41.x86_64
ls: Zugriff auf '/usr/src/kernels/6.11.4-301.fc41.x86_64' nicht möglich: Datei oder Verzeichnis nicht gefunden

csg@fedora:~/git/hid-fanatecff$ sudo ls -la /usr/src/kernels/6.11.4-301.fc41.x86_64
[sudo] Passwort für csg: 
ls: Zugriff auf '/usr/src/kernels/6.11.4-301.fc41.x86_64' nicht möglich: Datei oder Verzeichnis nicht gefunden

csg@fedora:~/git/hid-fanatecff$ sudo ls -la /usr/src/kernels
insgesamt 0
drwxr-xr-x. 1 root root  46 20. Apr 21:08 .
drwxr-xr-x. 1 root root  24 24. Okt 16:49 ..
drwxr-xr-x. 1 root root 404 20. Apr 21:09 6.13.11-200.fc41.x86_64
```

samowitsch, 2025-04-21:
I am not that linux pro but GRUB was loading an older kernel and the grubby command
```bash
csg@fedora:~$ sudo grubby --set-default /boot/vmlinuz-6.14.2-300.fc42.x86_64
```  
helped me to fix this.

No i can compile the Fanatec kernel extension on Fedora Linux 42 (Workstation Edition) and it will be copied to the desired location but it still not work:
```bash
csg@fedora:~$ uname -r
6.14.2-300.fc42.x86_64
csg@fedora:~$ cd git/hid-fanatecff/
csg@fedora:~/git/hid-fanatecff$ make

:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Verzeichnis „/usr/src/kernels/6.14.2-300.fc42.x86_64“ wird betreten
make[2]: Verzeichnis „/home/csg/git/hid-fanatecff“ wird betreten
make[2]: Verzeichnis „/home/csg/git/hid-fanatecff“ wird verlassen
make[1]: Verzeichnis „/usr/src/kernels/6.14.2-300.fc42.x86_64“ wird verlassen
csg@fedora:~/git/hid-fanatecff$ sudo make install
[sudo] Passwort für csg: 

:: Installing Fanatec kernel module/udev rule
=====================================================
'hid-fanatec.ko' -> '/lib/modules/6.14.2-300.fc42.x86_64/kernel/drivers/hid/hid-fanatec.ko'
'fanatec.rules' -> '/etc/udev/rules.d/99-fanatec.rules'
depmod
csg@fedora:~/git/hid-fanatecff$ sudo udevadm control --reload-rules && sudo udevadm trigger
csg@fedora:~/git/hid-fanatecff$ lsmod | grep fanatec
csg@fedora:~/git/hid-fanatecff$ lsmod | grep hid
hid_apple              28672  0
```

I don't know what to do yet 🤷 It seems that the kernel extension are now .xz compressed.

```bash
csg@fedora:~$ ll /lib/modules/6.14.2-300.fc42.x86_64/kernel/drivers/hid/
insgesamt 2836
...
...
-rw-r--r--. 1 root root    2964 10. Apr 02:00 hid-ezkey.ko.xz
-rw-r--r--. 1 root root 1494160 21. Apr 19:02 hid-fanatec.ko
-rw-r--r--. 1 root root  288128 21. Apr 00:27 hid-fanatec.ko.xz
...
...
```

gotzl, 2025-04-22:
Did you do `sudo dnf install kernel-devel` before `make`ing the driver?

andreasntr, 2025-07-23:
> Did you do `sudo dnf install kernel-devel` before `make`ing the driver?

i have the same issue, i can confirm i run all the steps from kernel-devel installation to udev reload

I also tried to start the games (ACC and AMS2) with proton GE 9-27. Buttons and pedals work fine but FFB is completely absent. As the other user, i also don't see the fanatec_hid when running `lsmod | grep hid`

gotzl, 2025-07-23:
Can you load the module manually with `insmod hid-fanatec.ko` after doing `make`? Then, plugin/power-on the wheel and post `dmesg | tail -n50`.

andreasntr, 2025-07-24:
I'm getting `insmod: ERROR: could not insert module hid-fanatec.ko: Key was rejected by service`

andreasntr, 2025-07-24:
It works now, but I had to make some research. I found out that on Fedora you either sign custom modules or you have to disable secure boot. I did the latter. Can you add it to the README somewhere?

Thank you