gotzl/hid-fanatecff #113: Permissions do not set on sysfs nodes sometimes

Spacefreak18, 2025-02-14:
Hello, I've been following your work for sometime and it's marvelous. Please keep it up!

I'm not sure what the issue is because on my development machine the sysfs are useable by my unprivileged user, yet it does not work on my production machine. Both are up to date Debian Testing.

On the machine where the permissions do not set correctly, if i run ```udevadm control --reload-rules && udevadm trigger``` after i'm logged in and notice I can't write to the sysfs rumble, then it will work. So it seems the rule is fine.

It must just be running before the sysfs node is actually created?

there used to be a WAIT_FOR_SYSFS key in udev, but it's been deprecated https://lists.freedesktop.org/archives/systemd-devel/2015-July/033485.html

What I ended up doing is i added a line to systemd-tmpfiles. (/etc/tmpfiles.d/tmp.conf)
```Z /sys/devices/pci0000:00/*/*/*/*/*0003:0EB7:183B*/* 1777 root games -```

I just wanted to get this written up and out to the wider internet, to see if anyone has any solutions, and just to raise awareness if anyone else has the same issue. Or knows what i'm doing wrong on one machine but not the other.

gotzl, 2025-06-07:
Can you the output of `udevadm test /dev/input/by-id/usb-Fanatec*-event-joystick` for the machine with permission problems?

Spacefreak18, 2025-11-17:
some combination of updates made it so i no longer need that tmpfiles workaround.

i'm closing this for you. Thank you for even commenting. :)