gotzl/hid-fanatecff #114: iRacing won't start when a Fanatec Wheel is present

mfoacs, 2025-03-23:
I've been playing around with iRacing on Steam which, with exception to the anti-chat not being implemented on Linux, works pretty well with minor tweaks out of the box.

One of them is starting a racing session with the wheel turned off. Then turning it on, mapping the inputs and hopefully, force feedback will work. Works 1 on 6 to 7 attempts. 

What never works, is connecting the wheel, turning it on and then launching the game. The session gets stuck at "Launching server for offline session..." until it times-out.

None of the suggested troubleshooting methods worked for me: flag, registry keys or input during game launch.

System specs:
`Linux nobara 6.13.7-200.nobara.fc41.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Mar 13 19:39:30 UTC 2025 x86_64 GNU/Linux`
`Steam Version:  1741737356`
`modinfo hid-fanatec | grep hidraw`
`#\> parm:           hidraw_pid:bool`

Wheelbase is Fanatec ClubSport DD+, Podium/BME, ClubSport Pedals V3.

I'm happy to keep trying different things if you point me in the right direction!

This is awesome btw, works flawlessly in other titles such as ACC, ACE and ExoCross.

mfoacs, 2025-03-27:
In case anyone is interested, I've found a workaround:
1. Add `PROTON_ENABLE_HIDRAW=1 %command%` on the properties of the game;
2. Connect the wheelbase but don't turn it on;
3. Run `protontricks -c "wine reg delete 'HKLM\System\CurrentControlSet\Enum\HID' /f"` before starting a session.
4. When the session is starting, and you see "Connected!" on the render progress bar, turn the wheelbase on and quickly run `sudo udevadm control --reload-rules && sudo udevadm trigger`;
5. Reconfigure the controls. If you see a message saying "Force feedback is not working properly" close the session, turn off the wheelbase and repeat 3, 4, 5.
Following these steps, I have managed to get it running consistently and with force feedback.

liszca, 2025-05-30:
iRacing doesn't run under Linux as Anti Cheat is not working. Thats what I heard.

mfoacs, 2025-05-30:
> iRacing doesn't run under Linux as Anti Cheat is not working. Thats what I heard.

iRacing runs under Linux, Anti Cheat does not. Which means, you can use it as long as you're not on an Official session. It's useful for practice sessions, reviewing replays and, most importantly, staying away from Windows...

mfoacs, 2025-05-30:
> Hi, can you retest iracing with driver version 0.2.0 and a recent ge-proton version? My expectation would be that no tricks are required anymore, the game should start with connected/powered wheel w/o issues.

Unfortunately, I see no change. In fact, if the wheelbase is turned on and connected before you click "Play" on Steam, the game won't start at all. Once the UI is up and running, I can turn the wheel on. Then, disconnect the USB cable, start a practice session and once the render engine has connected, ("Connected!") plug the wheelbase back on, and you're ready to go.

gotzl, 2025-06-04:
Hmm.. thanks again for testing, that's super weird ^^
Could you try clearing enumerated HID devices like described in [the readme](https://github.com/gotzl/hid-fanatecff?tab=readme-ov-file#game-hangscrashes-at-startup)? Maybe that helps ...

mfoacs, 2025-06-25:
I reinstalled my OS and I changed from Ubuntu to OpenSuSE Tumbleweed. The
issue is gone, it works well now.

gotzl, 2025-06-25:
It does? Do you also get FanatecSDK functions, like LED / display? I then might add it to the list of confirmed working games and close this ticket...

closed 2026-10-18 by gotzl