gotzl/hid-fanatecff #116: FFB does not work properly in ISIEngine Games

its-just-michael, 2025-04-12:
I have tried 3 different ISIEngine games so far, GTR2, F1 Challenge 99-02, and AMS1. All of them have the exact same behavior. They detect my wheel and take its input just fine, and certain FFB effects (Curb, Crashing, etc) work like you'd expect them to. However, no matter what FFB settings I change and fiddle with, the normal steering forces will not work. I feel absolutely nothing in the wheel until I hit something. Tried numerous different Proton/Wine versions without it making a difference. I'm using Debian 12 Bookworm and a CSL Elite Wheelbase. FFB works 100% fine in pretty much every other game.

gotzl, 2025-07-06:
Hi,
just bought AMS1. With `PROTON_DISABLE_HIDRAW=1 %command%` I at least get curb and crash effects, but still no FFB from tyres. I've to investigate ..

megamster, 2025-08-01:
Have that same issue with a Logitech G25 wheel. Using Proton 6 fixes it. Seems Proton 7+ breaks  compatibility with those games

MmAaXx500, 2026-08-19:
Crosslinking this issue from hid-tmff2 as it seems very similar to me https://github.com/Kimplul/hid-tmff2/issues/207

TheIronWolfModding, 2026-09-10:
This branch fixes canned effects: https://github.com/TheIronWolfModding/hid-fanatecff/tree/w-rumble-stuck .  I only tested GTR2 and DD1 wheel.  Enjoy!