gotzl/hid-fanatecff #121: Fails to build for Kernel 6.15

A1ic3-g, 2025-05-30:
Fails to build for kernel 6.15.

Seems like hrtimers_init has been deprecated. 

Same issue in another project: https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg2039766.html

```
❯ make CC=clang

:: Compiling Fanatec kernel module
========================================
make -C /lib/modules/`uname -r`/build M=$PWD
make[1]: Entering directory '/usr/lib/modules/6.15.0-2-cachyos/build'
make[2]: Entering directory '/home/alice/hid-fanatecff'
  CC [M]  hid-ftecff.o
hid-ftecff.c:1139:2: error: call to undeclared function 'hrtimer_init'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
 1139 |         hrtimer_init(&drv_data->hrtimer, CLOCK_MONOTONIC, HRTIMER_MODE_REL);
      |         ^
hid-ftecff.c:1139:2: note: did you mean 'hrtimers_init'?
/usr/lib/modules/6.15.0-2-cachyos/build/include/linux/hrtimer.h:387:20: note: 'hrtimers_init' declared here
  387 | extern void __init hrtimers_init(void);
      |                    ^
1 error generated.
make[4]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/scripts/Makefile.build:208: hid-ftecff.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/Makefile:2039: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/home/alice/hid-fanatecff'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.0-2-cachyos/build'
make: *** [Makefile:8: default] Error 2
```

A1ic3-g, 2025-05-30:
There seems to be a function `hrtimers_init` but it has a different number of arguments

gotzl, 2025-05-30:
I've just merged the PR, thanks for reporting/testing!

closed 2025-05-30 by gotzl