gotzl/hid-fanatecff #124: ACC force feedback sends the well to the far left

GoLD-ReaVeR, 2025-05-30:
Hey, I don't know if this is a me problem or something with the driver, but ACC when starting a practice session sends the wheel to the far left with maximum force. I didn't have this problem the previous time I tried using the wheel.

gotzl, 2025-06-04:
Ouch ;) I've not experienced s.t. like that so far. Did you only test ACC? Does it work in other games? Which driver / proton version are you using?

GoLD-ReaVeR, 2025-06-05:
I tried starting Assetto Corsa but that refuses to start for some reason, I could perhaps try a force feedback tester... I'm using the aur dkms version which should be the latest git version from here and the TKG build compiled a few days ago.

GoLD-ReaVeR, 2025-06-05:
ffmvforce shows very eratic behavior, the center seems to be somewhere at around +50% but the wheel seems to move in different directions along the line.

gotzl, 2025-06-07:
> the center seems to be somewhere at around +50%

What exactly do you mean with that?

Can you post the output of `evdev-joystick -s /dev/input/by-id/usb-Fanatec*-event-joystick` and check the values you get in `evtest` around the wheel center position?

GoLD-ReaVeR, 2025-06-07:
zsh 3921 [1] % evdev-joystick -s /dev/input/by-id/usb-Fanatec*-event-joystick
Supported Absolute axes:
  Absolute axis 0x00 (0) (X Axis) (value: 32473, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x01 (1) (Y Axis) (value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x02 (2) (Z Axis) (value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x03 (3) (X Rate Axis) (value: 0, min: -128, max: 128, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x04 (4) (Y Rate Axis) (value: 0, min: -128, max: 128, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x05 (5) (Z Rate Axis) (value: 65535, min: 0, max: 65535, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x06 (6) (Throttle) (value: 255, min: 0, max: 255, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x07 (7) (Rudder) (value: 0, min: -128, max: 127, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x10 (16) (Hat zero, x axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x11 (17) (Hat zero, y axis) (value: 0, min: -1, max: 1, flatness: 0 (=0.00%), fuzz: 0)
  Absolute axis 0x28 (40) (Miscellaneous) (value: 3, min: 0, max: 255, flatness: 0 (=0.00%), fuzz: 0)

Event: time 1749318435.880011, -------------- SYN_REPORT ------------
Event: time 1749318435.902016, type 3 (EV_ABS), code 0 (ABS_X), value 32557

gotzl, 2025-06-07:
Hmm... Can you show me the terminal output of `ffmvforce` for the position where it gives no force? Also, which device is it?

GoLD-ReaVeR, 2025-06-07:
Fanatec DD2 with Formula 2.5x wheel iirc.

As for the position where it gives no force:
mouse: 302 205 n: 0.51 0.03 angle: 1.62
level: 4147 direction: c1fe

And the location was difficult to find even. Any other location it goes max force in either direction.

gotzl, 2025-06-07:
Are you using the latest Fanatec firmware? If not, then I have an idea ...

GoLD-ReaVeR, 2025-06-08:
I tried updating the firmware, but after finally succeeding in updating the base firmware the fanatec software doesn't recognize the device anymore. As I've reached the end of my patience with this 1500 euro fiasco I'll try again later, possibly levy some words at fanatec support.

GoLD-ReaVeR, 2025-06-11:
This firmware updater was made by a buffoon. It doesn't actually work and when you force it you get yourself into more trouble. This is the most ridiculous thing ever for a company that is advertising themselves to the level they do and selling equipment at the prices they do. I've been able to update to the 440 driver but nothing beyond that. The game doesn't seem to recognize the wheel anymore but Linux still does. This is just a mess really. If you could make something that would allow us to force flash the device(s) that would be great.