gotzl/hid-fanatecff #128: clubsport V3 pedals only

fillman86, 2025-06-17:
I have a miss-match of things, one of which is clubsport V3 pedals (with break vibration motor) connected directly to a generic unpowered usb hub, then to my PC. I almost exclusively use BeamNG's native linux version, and it won't detect the pedals. After installing this, I tried proton and it worked, not sure if it needed the driver or not, but proton's version of beamng has problems, so I want to use them in the native version.

fillman86, 2025-06-26:
thanks. It picks up my moza wheel and handbrake, my generic shifter, and all controllers I have.

I have done a fair chunk of digging, and I believe it has to do with the pedals not having any buttons. I don't know the particulars, but apparently linux has some kind of issue with inputs that don't have buttons, or the type of electronics that don't have buttons. However I'm a fairly well known beamng modder, and know where the game files that look at controllers are, but I'm a complete novice at anything outside of beamng and its lua code. I've never looked at any code of anything else, like linux/drivers/github repositories at all. So if you could tell me what things to google, I might be able to start learning things, and figure out how to fix it myself :)

irseny, 2025-08-09:
Hi @fillman86,
It seems you're dealing with input device detection issues, which I've addressed with a compatibility tool [protopedal](https://gitlab.com/openirseny/protopedal).<br>
If you're still seeking a linux solution, you might consider trying the most basic case:<br>
`protopedal --button B /dev/input/by-id/usb-<your-device-here>-event-joystick`