gotzl/hid-fanatecff #131: [Issue] 2520º after connect the wheel base to the pc

davidm2m, 2025-08-21:
Hello,
I’m experiencing an issue when using my ClubSport DD+ wheel on Linux with the module. The behavior is as follows:

When the wheel is connected to Linux right after finishing its startup process, it sets the rotation to 25020 degrees, but the actual rotation that gets registered is as if it were 1080 degrees. For example, if you turn the wheel 90 degrees, only about 33 degrees are registered in the games.

If you don’t power off the wheel and then connect it to Windows, it continues to behave the same way: turning the wheel 90 degrees only registers about 33 degrees in the Fanatec software. However, if the wheel is powered off and restarted, it works correctly.

I also noticed that when using Assetto Corsa with Content Manager, this problem is fixed after loading any session with any car, as long as the option “auto match scale to match car direction” is enabled in the wheel configuration.

The SENSITIVITY setting on the wheel is set to AUTO. If I manually set it to 1080 degrees, there are no issues, but then the automatic adjustment per car (handled by the games) is lost.

I’m using Arch Linux with kernel 6.16.1, but this issue has been present for some time. I can’t tell exactly in which version it started happening.

Steps to reproduce:

Power on the wheel and connect it to the PC (I usually leave the USB connected, so it auto-connects right after the startup process).

Check the steering angle. I usually test it in a game other than Assetto Corsa with Content Manager. For example, in AMS2 it’s easy to confirm the issue described above.

To fix the problem, open Assetto Corsa with Content Manager and make sure the option “auto adjust scale to match car direction” is enabled.

If you need any additional information or logs, please let me know (and how to extract them), and I will add them to this issue.

odinu, 2026-10-01:
Thanks, I found your fix useful!
Also noticed the following, mentioning them in case anyone else has the same issues:
1. unloading the driver gives me a Linux BSOD 😱 
2. init_range doesn't appear in `/sys/module/hid_fanatec/parameters`, but it does work

gotzl, 2026-10-14:
Had no time to look close into the original issue (seems the workaround is sufficient for now), but I'm interested in the BSOD @odinu . This would be a high prio issue... Can you give logs/stacktrace?
Also `init_range` does not appear because only parameters with description (`MODULE_PARAM_DESC` in code) seem to appear there. The full list of params you can see with `modinfo hid_fanatec`. (I'm going to add descriptions for the other params as well).