gotzl/hid-fanatecff #132: [Issue] JDM: Japanese Drift Master ffb problem

davidm2m, 2025-08-21:
Hi,
I’ve been trying to play JDM: Japanese Drift Master on Linux with Proton using my ClubSport DD+ wheel, but I’m having problems with the force feedback.

On Windows the wheel works fine and the force feedback behaves normally, no issues at all.

On Linux, with Proton GE 10.11 (and also with some earlier versions), the problem appears. The wheel is detected correctly, I can configure it without issues, and by default it even loads a preset profile for the GT DD Pro.

The issue is that once I’m in the game, if I move the wheel just a little bit, it suddenly snaps hard all the way to the left. If I turn it slightly to the right, it stops spinning left, but as soon as I turn it even a little bit to the left again, it snaps left very abruptly until it reaches the end of the wheel rotation.

Steps to reproduce:

Connect the ClubSport DD+ wheel to the PC running Linux.

Launch the game with Proton GE 10.11.

Configure the wheel in the game settings (recognized correctly).

Start a race.

Slightly turn the wheel left or right.

Expected behavior:
Force feedback should behave normally, without unexpected snapping.

Actual behavior:
When slightly turning the wheel left, the wheel suddenly snaps all the way to the left until it reaches the rotation limit.

System info:

Distro: Arch Linux

Kernel: 6.16.1 (also happened on earlier versions)

Proton: GE-Proton 10-11

gotzl, 2025-10-04:
Hi,
thanks for the report. I just bought the game on sale and can confirm the issue. I think the unexpected snapping goes away my using `PROTON_DISABLE_HIDRAW=1 %command%`, which means the problem is inside my HID PID code.
However, even with setting this, I can't play the game properly because all axis inputs behave binary (eg. steering slightly left (right) registers in-game as full left (right)). I've to investigate some more to figure this out ....

davidm2m, 2026-01-06:
Hello.
I have installed the devel branch and tested it in JDM, and everything seems to be working correctly now. In my case, the pedals connected to the CS DD+ base work without any issues and are correctly recognized as axes.

As for other games, I haven’t encountered any problems when using the devel branch.

All tests were made with hidraw enabled.